blinker-iot/blinker-library #287: Please stop stealing Blynk intelectual property

vshymanskyy, 2018-12-22:
This library is clearly a derivative of https://github.com/blynkkk/blynk-library
According to Blynk Library license, this derivative work should have proper attribution to the original product.
Please resolve this ASAP, or we will resort to law enforcement mechanisms.

i3water, 2018-12-23:
Dear blynk founder, thanks your groups work first.
Blinker seems like blynk in first look, but not a derivative of blynk.
To be honest, I had forked blynk codes before, and I use the template class(c++) same as blynk to build blinker.
But the way how to communicate and connect server and so on are all different with blynk.

i3water, 2018-12-23:
> Well, there is no difference "how much" you change the code. The fact you used our library as a base, renamed everything and removed the license - is a rude violation of intellectual property rights. Please read Blynk library license carefully.
> […](#)

Dear vshymanskyy,
I forked your codes not means I changed your codes. Thanks.
if you see some codes change or used your codes, let me know, I will delete it.

vshymanskyy, 2018-12-23:
So you took our library, copied it structure, architecture, some of it's unique properties, even some of the code (with proper renaming) - and now claim it is not a derivative work.
Again, please respect the license and IP rights.

## Project structure - the same.
[image]
___
## Architecture - the same.
[image]
___
## Logos - the same.
[image]
___
## Configuration options - the same.
[image]
___
## Every example...
Contains a very similar header and description. Come on!
[image]

There are MANY, MANY places like this.
We have reported this to github, as a first counter-measure.

EvilPlus, 2018-12-23:
Are u silly B ??? @vshymanskyy . all your comments  are Force Bulls*** . I can't see any plagiarism in the code.

But i see u dont have mind.

vshymanskyy, 2018-12-23:
How sweet, EvilPlus:
[image]

coloz, 2018-12-23:
about structure,this is a common network connection c++ structure
about logo,this is the software,http://patorjk.com/software/taag/，automatically generated
about options,don‘t you think it's normal to use macro definitions?
about example header and description， we'll rewrite it to avoid your misunderstanding

coloz, 2018-12-23:
@vshymanskyy 
We are improving a smart home solution.This library is part of it,maybe there are some overlapping functions with your project
If you read the library code, you will find that we did not use your code

vshymanskyy, 2018-12-23:
> about structure,this is a common network connection c++ structure

If it is "common", please find ANY project that uses this particular approach.

> about logo,this is the software,http://patorjk.com/software/taag/ automatically generated

Sure, all logos are basically generated in some software nowadays. With some human help.

> about options, don‘t you think it's normal to use macro definitions?

Macro definitions are OK, but specific names **DO remind me** exactly the same in Blynk.
This particular place is very simple, but most probably copy-pasted.

> about example header and description， we'll rewrite it to avoid your misunderstanding

Thanks.

coloz, 2018-12-23:
@vshymanskyy 
I don't know where it's special.You can read more  programs about iot and arduino.
copy-pasted?You're right about The definition file is too simple，so,No need to use your code

vshymanskyy, 2018-12-23:
@i3water All of the projects you mentioned, use a different architecture approach (inheritance, virtual functions, etc.). Please check their code carefully.

vshymanskyy, 2018-12-23:
Guys, sorry but I'm leaving conversation here.
I have no direct interest in discussing it with you.
Thanks for understanding.

coloz, 2018-12-23:
@vshymanskyy Please tell us the same code. It seems that you just think we should not use the correct C++ grammar.

vshymanskyy, 2018-12-23:
It does not mean, the issue is resolved ;)

vshymanskyy, 2018-12-23:
@coloz please consult with your lawyer.

coloz, 2018-12-23:
@vshymanskyy 
It's all your arbitrary guesses.

i3water, 2018-12-23:
> @i3water All of the projects you mentioned, use a different architecture approach (inheritance, virtual functions, etc, with a completely different structure). Please check their code carefully.

oh no, with my poor english. I dont konw you talk about project or codes structure, what are you talk about? You mean template class ? I told before , I used the same template class(always some inheritance) to buid blinker.

coloz, 2018-12-23:
@vshymanskyy 
用中文回复下，避免英语不好造成的表达错误。i3water说，fork过你的代码，是说他的账号fork过，并不是这个项目使用了你的代码。
如果您阅读了我们的代码，您会发现我们使用了不一样的结构，不一样的代码，不一样的底层库，并且在说明文档中我们注明了使用了的库的链接。
我们从一开始就注意过各个项目开源协议，并且在合法的前提下使用这些项目。您不能因为我们项目和您的项目存在相似的功能，就判断这个库派生自您的项目。
对于名称的类似，完全是个意外，对此表示抱歉，我一开始并不知道blynk的读音也是blink，我们取名叫blinker，是因为我们中文名叫点灯，翻译过来就是blink，上个月我才知道blynk也是这么读。
如果名称让您不高兴，我们暂时也没办法更改，因为在中国审核软件是件很麻烦的事。

i3water, 2018-12-25:
@vshymanskyy 
We had upgrade codes and use new architecture approach.
Although we dont think blinker is a derivative of blynk, we thanks your works and differing opinion.

i3water, 2018-12-25:
@vshymanskyy 
If you dont have other opinions I will close this issue next monday, thanks.

vshymanskyy, 2018-12-25:
@i3water I'm pretty satisfied with the architecture changes you made.
I also example headers updated, thanks!
One specific request I have, is also to remove the ascii text banner, which also resembles that of Blynk too much:
[image]

I'll also do one more review of several parts (that I didn't have to check out yet), and I'm sure we can resolve it by Monday.

coloz, 2018-12-26:
@vshymanskyy 
我们拒绝这个更改，这仅仅是个自动生成的标题而已，如果该字体侵犯了您的权利，您可以投诉这个字体生成软件。重申，blinker不是blynk的衍生版，虽然我对名字读音的类似表示抱歉，但我们有对自己项目命名的自由。

i3water, 2018-12-26:
@vshymanskyy 
I am sorry about this banner, we also need a banner to display our LOGO/VERSION and some prompt information. If you think the information codes resembles you, we will change the information codes, but we may not remove this banner.

kasuganosoras, 2018-12-26:
> @vshymanskyy
> I am sorry about this banner, we also need a banner to display our LOGO/VERSION and some prompt information. If you think the information codes resembles you, we will change the information codes, but we may not remove this banner.

不就是个 figlet 生成的字符画么……我也喜欢给自己软件启动的时候命令行输出个字符画
要别人删除这个简直是无理取闹

bumfo, 2018-12-26:
> @i3water I'm pretty satisfied with the architecture changes you made.
> I also example headers updated, thanks!
> One specific request I have, is also to remove the ascii text banner, which also [...]

Although the idea of using ASCII Art as logo makes it resemble your project, this is not a violation of copyright, since ideas cannot be copyrighted. You can resubmit this request once your project is patented, thanks.

closed 2018-12-31 by i3water