blinker-iot/blinker-library #550: Arduino中ESP8266开发板更新最新2.7.3版本，程序加入ESPTOUCH配网失败

molun, 2020-08-02:
## 设备信息
- 硬件: [ESP-12F、ESP-01S、NodeMCU]
- 接入方式: [WIFI]
- package版本: [esp8266-arduino-2.7.3]
- blinker库版本: [arduino 0.3.4]
- 开发环境: Arduino
- 操作系统:Windows

## IDE 中的设置
- Module: [Nodemcu]
- Flash Size: [4MB]
- lwip Variant: [v2 Lower Memory]
- Reset Method: [ck|nodemcu]
- Flash Frequency: [40Mhz]
- CPU Frequency: [80Mhz]
- Upload Using: [OTA|SERIAL]
- Upload Speed: [115200] 
[image]

## 手机信息及APP信息
- 手机品牌及型号 ：小米6
- android系统版本：9
- blinker App版本：2.4.6

## 问题描述
Arduino中ESP8266开发板更新最新2.7.3，刷写编译ESPTOUCH配网信息后，在串口监视器已经查看到成功获取到IP地址，还是不断在循环获取配网信息，blinker APP 中设备并不在线，配网也搜索不到设备。网络连接信息应该是之前连接过的缘故吧！ESP-12F、ESP-01S、NodeMCU都存在以上问题，退回2.7.2版本问题消失

i3water, 2020-08-02:
非blinker lib的原因，2.7.3版本package中对https client支持库出现错误。已向ESP8266/ARDUINO提交ISSUE，待官方更新。
目前可以回退到2.7.2版本即可。

i3water, 2020-08-03:
官方已修复并更新2.7.4版本package，更新即可。

closed 2020-08-03 by i3water