blinker-iot/blinker-library #590: blinker 老版本使用阿里云中介服务的 MQTT 协议 RRPC 具体订阅规则谁知道

blinker123, 2022-03-26:
blinker 如果使用 中介是blinker 自身，则使用 /device/${deviceName}/[r|s] 可以正常[发送|接收]消息，并不需要额外代码，普通 attach就可以读取到内容，如果中介使用 阿里云，则需要 准守 rrpc  发送规则，还要看blinker 与阿里云amqp 是怎么样定义的，

esp8266 与阿里云 /sys/${productKey}/${deviceName}/rrpc/request/  rrpc 订阅
esp8266 与BLinker  /device/${deviceName}/r 订阅  

什么样规则的 TOpic 可以发送一条rrpc 消息到 设备上呢。

i3water, 2022-03-26:
rrpc发布消息并不需要什么规则，满足阿里rrpc发布要求即可。只不过普通用户也无法通过阿里接口发送rrpc topic数据，所以不做更多实现代码及方案的讲解。

closed 2022-03-26 by i3water

blinker123, 2022-03-26:
> rrpc发布消息并不需要什么规则，满足阿里rrpc发布要求即可。只不过普通用户也无法通过阿里接口发送rrpc topic数据，所以不做更多实现代码及方案的讲解。

多谢回复，使用阿里云的设备，通过mqtt 是可以收到  binlinker 官方控制台调试收到的消息，我就是想实现这个接收功能，当然发送 肯定要rrpc 发送，

Topic: /JgCGbHlndgz/7F75A483KY164CXQHCGS6N41/rQoS: 0
{"deviceType":"DiyArduino","data":{"get":"state"},"fromDevice":"05d561be755e1ee2f06b122a223e5e13","toDevice":"7F75A483KY164CXQHCGS6N41"}

这是设备收到的 mqtt 消息，我就想自己发送 这样的消息，不需要点灯设备里面的调试来发送

i3water, 2022-03-26:
这个消息及topic并不是rrpc方法实现的也不是rrpc topic，就是常规的mqtt消息发布订阅而已，参考对应nqtt broker提供的文档即可。由于这些问题与blinker不相关，可能后面不会再次回复该问题。