blinker-iot/blinker-library #593: 合宙ESP32C3无法通过编译

pdtopdog, 2022-07-27:
## 设备信息
- 硬件: ESP32C3
- 接入方式: WIFI
- package版本: esp32-2.0.4
- blinker库版本: arduino 4.0
- 开发环境: Arduino
- 操作系统: Windows

## IDE 中的设置
- Module: SP32C3 DEV
- Flash Mode: DIO
- Flash Size: 4MB
- Flash Frequency: 80Mhz
- CPU Frequency: 160MHz
- Upload Using: SERIAL
- Upload Speed: 115200

## 问题描述
不能编译

## 代码
/* *****************************************************************
 *
 * Download latest Blinker library here:
 * https://github.com/blinker-iot/blinker-library/archive/master.zip
 * 
 * 
 * Blinker is a cross-hardware, cross-platform solution for the IoT. 
 * It provides APP, device and server support, 
 * and uses public cloud services for data transmission and storage.
 * It can be used in smart home, data monitoring and other fields 
 * to help users build Internet of Things projects better and faster.
 * 
 * Make sure installed 2.7.4 or later ESP8266/Arduino package,
 * if use ESP8266 with Blinker.
 * https://github.com/esp8266/Arduino/releases
 * 
 * Make sure installed 1.0.5 or later ESP32/Arduino package,
 * if use ESP32 with Blinker.
 * https://github.com/espressif/arduino-esp32/releases
 * 
 * Docs: https://diandeng.tech/doc
 *       
 * 
 * *****************************************************************
 * 
 * Blinker 库下载地址:
 * https://github.com/blinker-iot/blinker-library/archive/master.zip
 * 
 * Blinker 是一套跨硬件、跨平台的物联网解决方案，提供APP端、设备端、
 * 服务器端支持，使用公有云服务进行数据传输存储。可用于智能家居、
 * 数据监测等领域，可以帮助用户更好更快地搭建物联网项目。
 * 
 * 如果使用 ESP8266 接入 Blinker,
 * 请确保安装了 2.7.4 或更新的 ESP8266/Arduino 支持包。
 * https://github.com/esp8266/Arduino/releases
 * 
 * 如果使用 ESP32 接入 Blinker,
 * 请确保安装了 1.0.5 或更新的 ESP32/Arduino 支持包。
 * https://github.com/espressif/arduino-esp32/releases
 * 
 * 文档: https://diandeng.tech/doc
 *       
 * 
 * *****************************************************************/

#define BLINKER_WIFI
#define BLINKER_WIDGET

#include <Blinker.h>

const int led = 4;
char auth[] = "********";
char ssid[] = "SSID";
char pswd[] = "PASSWORD";

BlinkerWiFi                 Blinker(WiFiESP);
// 新建组件对象
BlinkerButton<BlinkerWiFi>  Button1(Blinker, "btn-abc");
BlinkerNumber<BlinkerWiFi>  Number1(Blinker, "num-abc");

int counter = 0;

// 按下按键即会执行该函数
void button1_callback(const String & state)
{
    BLINKER_LOG("get button state: ", state);
    digitalWrite(led, !digitalRead(led));
}

// 如果未绑定的组件被触发，则会执行其中内容
void dataRead(const String & data)
{
    BLINKER_LOG("Blinker readString: ", data);
    counter++;
    Number1.print(counter);
}

void setup()
{
    // 初始化串口
    Serial.begin(115200);
    BLINKER_DEBUG.stream(Serial);
    BLINKER_DEBUG.debugAll();
    
    // 初始化有LED的IO
    pinMode(led, OUTPUT);
    digitalWrite(led, HIGH);
    // 初始化blinker
    Blinker.begin(auth, ssid, pswd);
    Blinker.attachData(dataRead);

    Button1.attach(button1_callback);
}

void loop()
{
    Blinker.run();
}

## LOG信息
无信息

## 编译信息
\blinker-library-dev_4.0\src/Blinker/BlinkerProtocol.h:261:64: error: 'class EspClass' has no member named 'getCoreVersion'; did you mean 'getSdkVersion'?
BLINKER_LOG(TAG_PROTO, BLINKER_F("package version: "), ESP.getCoreVersion());
^~~~~~~~~~~~~~
getSdkVersion
exit status 1

coloz, 2022-09-23:
请使用最新的dev4.0配合esp32 package 2.0.5测试

luqizhi666, 2023-02-26:
同问，请问解决了吗

coloz, 2023-02-28:
> 同问，请问解决了吗

一直是可用的，请确保您的esp32 package版本是最新的

closed 2023-02-28 by coloz