blinker-iot/blinker-library #598: Arduino编译有大量警告，因为不符合iso C++标准

flywhc, 2023-03-18:
## 问题描述
使用Arduino 1.8.19编译Hello Wifi有大量警告，
经查看这里的源码发现是因为错误声明导致的。
例如 
`class BlinkerButton:: BlinkerButton(char _name[], blinker_callback_with_string_arg_t _func = NULL)`

这里不应该使用数组而是应该使用const char *，才能方便调用者传入各种类型字符串。

还有错误使用register关键字，见后面编译提示。

## 代码
见官方范例 Hello WiFi

## 编译信息
```
\Blinker\examples\Blinker_Hello\Hello_WiFi\Hello_WiFi.ino:55:23: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
   55 | BlinkerButton Button1("btn-abc");
arduino\libraries\blinker\src\modules\taskscheduler\taskscheduler.h:872:28: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
  872 |     register unsigned long m, i;  // millis, interval;
```

coloz, 2023-03-26:
一个编程的基本常识：warning不是报错，通常都可以忽略，几乎您使用任何库都会有warning

closed 2023-03-26 by coloz

sherlocknieh, 2024-12-14:
可以修改编译参数屏蔽警告. 如果在用 PlatformIO IDE, 可以在 platformio.ini 里添加:

build_unflags = -Wwrite-strings
build_flags = -Wno-write-strings

第一行取消 write-strings 警告, 第二行强制关闭 write-strings 警告