blinker-iot/blinker-library #601: Blinker.attachRTData(rtData) 内核崩溃

Tonnp, 2023-05-14:
== 提问请到论坛 https://www.arduino.cn/forum-132-1.html ==

## 基本信息
- [] 这个问题满足提问指南
- [] 我阅读了开发文档
- [] 我使用了最新的库及APP进行测试，但是问题仍然存在
- [] 我搜索过类似问题的解决方法但是仍然无法解决
- [] 有设备端的LOG信息
- [] 有编译报错信息
- [] 我已填写基本信息

## 设备信息
- 硬件: [Arduino by ESP32-WROOM]
- 接入方式: [WIFI]
- package版本: [esp32-v3.0-dev]
- blinker库版本: [arduino v0.3.100221211]
- 开发环境: [Arduino]
- 操作系统: [Windows]

## IDE 中的设置
- Module: [Generic ESP8266 Module|Wemos D1 mini r2|Nodemcu|other]
- Flash Mode: [qio|dio|other]
- Flash Size: [4MB/1MB]
- lwip Variant: [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method: [ck|nodemcu]
- Flash Frequency: [40Mhz]
- CPU Frequency: [80Mhz|160MHz]
- Upload Using: [OTA|SERIAL]
- Upload Speed: [115200|other] (serial upload only)
## 手机信息及APP信息
- 手机品牌及型号  
- android系统版本（请注意是android系统版本，不是厂商自己的版本号）
- blinker App版本  

## 问题描述
使用实时数据，并使用RTData，遇见3个问题
1、数据组件少时未出现与MQTT服务器断开的情况（大概6个组件），随着数量增加，发送提示断开的概率会随之增加（MQTT Disconnected）
2、提示断开后会再次尝试重发，发送数据会在原来的发送data外在增加一层data，变成了  {"data":{"data":{内容}发送类型等}发送类型等}
      尝试使用  if(!STRING_contains_string(data,"data")){} 修复后没有2层data，但是还是会崩溃
3、随后就会出现内核崩溃
CORRUPT HEAP: Bad head at 0x3ffdb070. Expected 0xabba1234 got 0x00000000
abort() was called at PC 0x400871a1 on core 1
## 代码
setup里的代码
  Blinker.begin(auth, ssid, pass);
  // BLINKER_TAST_INIT();  //以为是线程导致的，屏蔽后依旧
  Blinker.attachData(dataRead);
  Blinker.attachHeartbeat(heartbeat);
  Blinker.attachRTData(rtData);

void rtData()
{
  Blinker.sendRtData("num-v1", ina1_voltage*1000);
  Blinker.sendRtData("num-v2", ina2_voltage*1000);
  Blinker.sendRtData("num-v3", ina3_voltage*1000);
  Blinker.sendRtData("num-i1", ina1_converted*1000);
  Blinker.sendRtData("num-i2", ina2_converted*1000);
  Blinker.sendRtData("num-i3", ina3_converted*1000);
  Blinker.sendRtData("num-t1", isnan(t1_temperature)?0:t1_temperature);
  Blinker.sendRtData("num-t2", isnan(t2_temperature)?0:t2_temperature);
  Blinker.sendRtData("num-t3", isnan(t3_temperature)?0:t3_temperature);
  Blinker.sendRtData("num-pwm1", ledcRead(pwmChannel_1));
  Blinker.sendRtData("num-pwm2", ledcRead(pwmChannel_2));
  Blinker.sendRtData("num-pwm3", ledcRead(pwmChannel_3));
  Blinker.printRtData();
}

## LOG信息
[22380] Got: {"fromDevice":"ee265a9ff0c911eda8eb5254","data":{"rt":["num-v1","num-i1","num-t1","num-pwm1","num-v2","num-t2","num-i2","num-pwm2","num-v3","num-t3","num-i3","num-pwm3"]}}
[22385] data: {"rt":["num-v1","num-i1","num-t1","num-pwm1","num-v2","num-t2","num-i2","num-pwm2","num-v3","num-t3","num-i3","num-pwm3"]}
[... 76 lines cut ...]
[24682] isJson: {"data":{"num-v1":{"date":1684025806,"val":0.00},"num-v2":{"date":1684025806,"val":1.25},"num-v3":{"date":1684025806,"val":0.00},"num-i1":{"date":1684025806,"val":-0.20},"num-i2":{"date":1684025806,"val":-0.20},"num-i3":{"date":1684025806,"val":-0.50},"num-t1":{"date":1684025806,"val":0.00},"num-t2":{"date":1684025806,"val":0.00},"num-t3":{"date":1684025806,"val":0.00},"num-pwm1":{"date":1684025806,"val":204},"num-pwm2":{"date":1684025806,"val":511},"num-pwm3":{"date":1684025806,"val":818}},"fromDevice":"121FD9E6BOYWRPESH8PX5KZY","toDevice":"ee265a9ff0c911eda8eb5254","deviceType":"OwnApp"}
[24736] MQTT Publish...
[24737] Freeheap: 202200
[24743] {"data":{"num-v1":{"date":1684025806,"val":0.00},"num-v2":{"date":1684025806,"val":1.25},"num-v3":{"date":1684025806,"val":0.00},"num-i1":{"date":1684025806,"val":-0.20},"num-i2":{"date":1684025806,"val":-0.20},"num-i3":{"date":1684025806,"val":-0.50},"num-t1":{"date":1684025806,"val":0.00},"num-t2":{"date":1684025806,"val":0.00},"num-t3":{"date":1684025806,"val":0.00},"num-pwm1":{"date":1684025806,"val":204},"num-pwm2":{"date":1684025806,"val":511},"num-pwm3":{"date":1684025806,"val":818}},"fromDevice":"121FD9E6BOYWRPESH8PX5KZY","toDevice":"ee265a9ff0c911eda8eb5254","deviceType":"OwnApp"}[24792] print: {"data":{"num-v1":{"date":1684025806,"val":0.00},"num-v2":{"date":1684025806,"val":1.25},"num-v3":{"date":1684025806,"val":0.00},"num-i1":{"date":1684025806,"val":-0.20},"num-i2":{"date":1684025806,"val":-0.20},"num-i3":{"date":1684025806,"val":-0.50},"num-t1":{"date":1684025806,"val":0.00},"num-t2":{"date":1684025806,"val":0.00},"num-t3":{"date":1684025806,"val":0.00},"num-pwm1":{"date":1684025806,"val":204},"num-pwm2":{"date":1684025806,"val":511},"num-pwm3":{"date":1684025806,"val":818}},"fromDevice":"121FD9E6BOYWRPESH8PX5KZY","toDevice":"ee265a9ff0c911eda8eb5254","deviceType":"OwnApp"}

[[24846] Proto print...
[24848] Freeheap: 24846] ...Failed
[24851] Freeheap: 245476
245476
[24854] [24854] Freeheap: 244304
isJson: {"data":{"data":{"num-v1":{"date":1684025806,"val":0.00},"num-v2":{"date":1684025806,"val":1.25},"num-v3":{"date":1684025806,"val":0.00},"num-i1":{"date":1684025806,"val":-0.20},"num-i2":{"date":1684025806,"val":-0.20},"num-i3":{"date":1684025806,"val":-0.50},"num-t1":{"date":1684025806,"val":0.00},"num-t2":{"date":1684025806,"val":0.00},"num-t3":{"date":1684025806,"val":0.00},"num-pwm1":{"date":1684025806,"val":204},"num-pwm2":{"date":1684025806,"val":511},"num-pwm3":{"date":1684025806,"val":818}},"fromDevice":"121FD9E6BOYWRPESH8PX5KZY","toDevice":"ee265a9ff0c911eda8eb5254","deviceType":"OwnApp"},"fromDevice":"121FD9E6BOYWRPESH8PX5KZY","toDevice":"ee265a9ff0c911eda8eb5254","deviceType":"OwnApp"}
[24919] MQTT Publish...
[24921] Freeheap: 246564
[24923] ERROR: MQTT Disconnected
[24926] Freeheap: 246564
CORRUPT HEAP: Bad head at 0x3ffdb288. Expected 0xabba1234 got 0x3ffdc814
abort() was called at PC 0x400871a1 on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x40088fd8:0x3ffb1ac0 0x40089255:0x3ffb1ae0 0x400871a1:0x3ffb1b00 0x400872cd:0x3ffb1b30 0x4010efc7:0x3ffb1b50 0x40108f89:0x3ffb1e10 0x40108efd:0x3ffb1e60 0x4008d969:0x3ffb1e90 0x400823ce:0x3ffb1eb0 0x40087099:0x3ffb1ed0 0x4000bec7:0x3ffb1ef0 0x400d452b:0x3ffb1f10 0x400e2d97:0x3ffb1f30 0x400e2e5d:0x3ffb1f90 0x400ee3dc:0x3ffb1fb0 0x4008a266:0x3ffb1fd0

Rebooting...

## 编译信息

Tonnp, 2023-05-14:
目前排查下来问题基本定位在Blinker.printRtData();
使用Blinker.print();做循环没有问题，然后rtdata会增加时间戳，不知是否json太长会导致服务器主动断开连接（json大概750字节）
另外Blinker.run()放在loop里，循环间隔大概是50ms左右，也会导致内核崩溃
ESP32 Guru Meditation Error: Core 0 panic'ed (LoadProhibited)

希望对排查有帮助

i3water, 2023-05-17:
从调试信息看可能是一次发送的数据太多导致堆栈不足，可以尝试减少一次发送的数据量进行测试。

QianZhongLi, 2025-05-06:
缩短键名为abcd单个字母