blinker-iot/blinker-library #604: ESP32 工程编译失败

ninstein, 2023-07-16:
== 提问请到论坛 https://www.arduino.cn/forum-132-1.html ==

## 基本信息
- [] 这个问题满足提问指南
- [] 我阅读了开发文档
- [] 我使用了最新的库及APP进行测试，但是问题仍然存在
- [] 我搜索过类似问题的解决方法但是仍然无法解决
- [] 有设备端的LOG信息
- [] 有编译报错信息
- [] 我已填写基本信息

## 设备信息
- 硬件: [ESP-32]
- 接入方式: [WIFI]
- package版本: [esp32 2.0.9]
- blinker库版本: [arduino 0.3.10]
- 开发环境: [Arduino]
- 操作系统: [Windows]

## IDE 中的设置
- Module: [ESP32 Dev Module]
- Flash Mode: [DIO]
- Flash Size: [4MB]
- Flash Frequency: [40Mhz]
- CPU Frequency: [80Mhz|160MHz]
- Upload Using: [SERIAL]
- Upload Speed: [115200] (serial upload only)

## 问题描述
选wifi连接demo工程编译失败，报错信息如下

```
In file included from C:\Users\abc\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.9\cores\esp32/Arduino.h:176,
                 from c:\Users\abc\Documents\Arduino\libraries\Blinker\src\modules\mqtt\Adafruit_MQTT.h:25,
                 from c:\Users\abc\Documents\Arduino\libraries\Blinker\src\modules\mqtt\Adafruit_MQTT.cpp:22:
c:\Users\abc\Documents\Arduino\libraries\Blinker\src\modules\mqtt\Adafruit_MQTT.cpp: In member function 'const char* Adafruit_MQTT::connectErrorString(int8_t)':
C:\Users\abc\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.9\cores\esp32/WString.h:38:55: error: cannot convert 'const __FlashStringHelper*' to 'const char*' in return
 #define F(string_literal) (FPSTR(PSTR(string_literal)))
                                                       ^
c:\Users\abc\Documents\Arduino\libraries\Blinker\src\modules\mqtt\Adafruit_MQTT.cpp:279:22: note: in expansion of macro 'F'
       case 1: return F("The Server does not support the level of the MQTT protocol requested");
                      ^
C:\Users\abc\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.9\cores\esp32/WString.h:38:55: error: cannot convert 'const __FlashStringHelper*' to 'const char*' in return
 #define F(string_literal) (FPSTR(PSTR(string_literal)))
                                                       ^
c:\Users\abc\Documents\Arduino\libraries\Blinker\src\modules\mqtt\Adafruit_MQTT.cpp:280:22: note: in expansion of macro 'F'
       case 2: return F("The Client identifier is correct UTF-8 but not allowed by the Server");
                      ^
C:\Users\abc\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.9\cores\esp32/WString.h:38:55: error: cannot convert 'const __FlashStringHelper*' to 'const char*' in return
 #define F(string_literal) (FPSTR(PSTR(string_literal)))
                                                       ^
c:\Users\abc\Documents\Arduino\libraries\Blinker\src\modules\mqtt\Adafruit_MQTT.cpp:281:22: note: in expansion of macro 'F'
       case 3: return F("The MQTT service is unavailable");
                      ^
C:\Users\abc\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.9\cores\esp32/WString.h:38:55: error: cannot convert 'const __FlashStringHelper*' to 'const char*' in return
 #define F(string_literal) (FPSTR(PSTR(string_literal)))
                                                       ^
c:\Users\abc\Documents\Arduino\libraries\Blinker\src\modules\mqtt\Adafruit_MQTT.cpp:282:22: note: in expansion of macro 'F'
       case 4: return F("The data in the user name or password is malformed");
                      ^
C:\Users\abc\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.9\cores\esp32/WString.h:38:55: error: cannot convert 'const __FlashStringHelper*' to 'const char*' in return
 #define F(string_literal) (FPSTR(PSTR(string_literal)))
                                                       ^
c:\Users\abc\Documents\Arduino\libraries\Blinker\src\modules\mqtt\Adafruit_MQTT.cpp:283:22: note: in expansion of macro 'F'
       case 5: return F("Not authorized to connect");
                      ^
C:\Users\abc\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.9\cores\esp32/WString.h:38:55: error: cannot convert 'const __FlashStringHelper*' to 'const char*' in return
 #define F(string_literal) (FPSTR(PSTR(string_literal)))
                                                       ^
c:\Users\abc\Documents\Arduino\libraries\Blinker\src\modules\mqtt\Adafruit_MQTT.cpp:284:22: note: in expansion of macro 'F'
       case 6: return F("Exceeded reconnect rate limit. Please try again later.");
                      ^
C:\Users\abc\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.9\cores\esp32/WString.h:38:55: error: cannot convert 'const __FlashStringHelper*' to 'const char*' in return
 #define F(string_literal) (FPSTR(PSTR(string_literal)))
                                                       ^
c:\Users\abc\Documents\Arduino\libraries\Blinker\src\modules\mqtt\Adafruit_MQTT.cpp:285:22: note: in expansion of macro 'F'
       case 7: return F("You have been banned from connecting. Please contact the MQTT server administrator for more details.");
                      ^
C:\Users\abc\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.9\cores\esp32/WString.h:38:55: error: cannot convert 'const __FlashStringHelper*' to 'const char*' in return
 #define F(string_literal) (FPSTR(PSTR(string_literal)))
                                                       ^
c:\Users\abc\Documents\Arduino\libraries\Blinker\src\modules\mqtt\Adafruit_MQTT.cpp:286:23: note: in expansion of macro 'F'
       case -1: return F("Connection failed");
                       ^
C:\Users\abc\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.9\cores\esp32/WString.h:38:55: error: cannot convert 'const __FlashStringHelper*' to 'const char*' in return
 #define F(string_literal) (FPSTR(PSTR(string_literal)))
                                                       ^
c:\Users\abc\Documents\Arduino\libraries\Blinker\src\modules\mqtt\Adafruit_MQTT.cpp:287:23: note: in expansion of macro 'F'
       case -2: return F("Failed to subscribe");
                       ^
C:\Users\abc\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.9\cores\esp32/WString.h:38:55: error: cannot convert 'const __FlashStringHelper*' to 'const char*' in return
 #define F(string_literal) (FPSTR(PSTR(string_literal)))
                                                       ^
c:\Users\abc\Documents\Arduino\libraries\Blinker\src\modules\mqtt\Adafruit_MQTT.cpp:288:23: note: in expansion of macro 'F'
       default: return F("Unknown error");
                       ^

exit status 1

Compilation error: exit status 1
```

ninstein, 2023-07-16:
我尝试手动修改Adafruit_MQTT的源代码，能编译通过，但是运行几小时左右板子会自动重启，曾经捕获到几次堆栈，堆栈信息如下：
```

Decoding stack results
0x400837e9: panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c line 408
0x4008d855: esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c line 137
0x400933f1: __assert_func at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/assert.c line 85
0x40093063: multi_heap_free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/heap/multi_heap_poisoning.c line 245
0x40083cad: heap_caps_free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/heap/heap_caps.c line 361
0x40093421: free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/heap.c line 39
0x400d4c8e: BLINKER_LOG_ALL__FlashStringHelper const*, String>(__FlashStringHelper const*, String) at c:\users\abc\documents\arduino\libraries\blinker\src\blinker/blinkerdebug.h line 101
0x400d4cd1: BLINKER_LOG_ALL__FlashStringHelper const*, String>(__FlashStringHelper const*, String) at c:\users\abc\documents\arduino\libraries\blinker\src\blinker/blinkerdebug.h line 101
0x400e428f: BlinkerOTA::update() at c:\users\abc\documents\arduino\libraries\blinker\src\blinker/blinkerdebug.h line 47
0x4011e01b: adjtime at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/time.c line 151
```

ninstein, 2023-07-16:
另一次异常重启的堆栈如下：
```
Decoding stack results
0x400837e9: panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c line 408
0x4008d855: esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c line 137
0x400933f1: __assert_func at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/assert.c line 85
0x40093063: multi_heap_free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/heap/multi_heap_poisoning.c line 245
0x40083cad: heap_caps_free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/heap/heap_caps.c line 361
0x40093421: free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/heap.c line 39
0x400d4c59: BLINKER_LOG_T__FlashStringHelper const*, String>(__FlashStringHelper const*, String) at c:\users\abc\documents\arduino\libraries\blinker\src\blinker/blinkerdebug.h line 47
0x400e5a5d: _lp_callback() at c:\Users\abc\Documents\Arduino\libraries\Blinker\src\Blinker\BlinkerTimer.cpp line 131
0x400e5c8a: BlinkerUpdaterClass::_writeBuffer() at c:\Users\abc\Documents\Arduino\libraries\Blinker\src\Functions\BlinkerUpdater.cpp line 1195
```

coloz, 2023-12-13:
经测试，使用github提供的库或者官网提供的库没有问题，不要通过pio或者arduino ide获取库，那不是最新版本

closed 2023-12-13 by coloz